ThunderFly-aerospace/TFRPM01 #3: Velikost otvoru pro matku sroubu

kaklik, 2020-06-26:
Sroub stahujici krabicku ma ve vicku sestihran pro zapusteni matice. Ten je ale moc maly a matice tam nejde vlozit.  
Ma o 0.6mm mensi prumer, nez by mela mit. 

Resenim zrejme je pouziti sroubu ze [standardni knihovny](https://github.com/ThunderFly-aerospace/Openscad_stdlib).

kaklik, 2020-12-09:
Pokusil jsem se použít knihovnu bolts.scad od @JanKott  Ale potíž je, že knihovna neumožňuje získat parametry rozměrů šroubu. Tudíž nelze její použití realizovat žádným přímým způsobem, neboť rozměry modelu jsou navázány na rozměry šroubu.

JanKott, 2020-12-09:
Upravim bolts knihovnu

JanKott, 2020-12-09:
Nyní jsou k dispozici funkce bolt_diameter, head_height, head_diameter, nut_height, nut_diameter, nut_pocket. Argumentem je velikost.

commit 8986cf2d (2020-12-09): Attempt to resolve #3.
Library is not actually used, because it does not have proper functionalities.
closed 2020-12-09 by JanKott