ThunderFly-aerospace/TFRPM01 #4: Rezistory v pouzdru 603

kaklik, 2020-09-21:
Rezitory R6, R8, R1 a R10. Případně také R7 a R5.  U kterých lze očekávat požadavky na změnu i po osazení modulu mají pouzdro 603. Které je citlivé na preciznost letování více než 805. 
Tato volba nedává smysl vzhledem k tomu, že v okolí těchto rezistorů je dostatek místa.

ChroustJan, 2021-02-17:
Navrhuji R7 nahradit pájecí ploškou. R1 a R10 dát do pouzdra 0805, zde opravdu hrozí změna uživatelem. R6 nahradit pájecí ploškou z výroby propojenou, která půjde přerušit nožem. A R8 nahradit pájecí ploškou.

kaklik, 2021-02-18:
Ok, přijde mi to jako rozumné řešení: 

[image]

closed 2021-02-18 by kaklik

kaklik, 2021-02-22:
Změna rezistoru R2 na pouzdo 805 má ještě jeden pozitivní přínos. 
A to ten že po vyřešení issue https://github.com/ThunderFly-aerospace/TFPROBE01/issues/5 na tomto rezitoru bude pravděpodobně trvale poněkud větší ztrátový výkon. Tím pádem je výhodné, že i rezistor má větší pouzdro.