ThunderFly-aerospace/TFRPM01 #7: Hall/Optical exteternal sensor board

kaklik, 2020-10-25:
Na základě některých [nových aplikací modulu](https://github.com/ThunderFly-aerospace/TF-G2/issues/64
) by zřejmě bylo vhodné vytvořit externí plošný spoj s magnetickým a optickým snímačem. 

Obdobně jako je: https://ozrobotics.com/shop/maleetronic-opb9000-reflective-optical-sensor/

### Reflexive sensor 

Od reflexivního snímače na plošném spoji potřebujeme, aby byl citlivý na otvory nebo prohlubně ve 3D vytištěném dílu. Kódovací kotouček by tak neměl vyžadovat žádnou dodatečnou povrchovou úpravu.

Alternativně by snímač měl být citlivý na kódovací pásky vytištěné na papír, nebo fólii. 

- [VCNT2020](https://www.vishay.com/optical-sensors/list/product-84285/) - zřejmě nejlepší možnost od VISHAY
- [OPTOI](https://optoi.com/en/products/components/reflective-sensors)

- [OPB9000](https://cz.mouser.com/ProductDetail/Optek-TT-Electronics/OPB9000?qs=Ok18xCFj7msWl1O2KsmQxQ%3D%3D)

### Hall magnetic sensor 

Magnetický snímač musí být citlivý na magnety zalepené do snímacího kotoučku. Použití magnetického snímače předpokládáme v prostředí, kde hrozí znečištění, které by mohlo ohrozit funkci optického snímače.

- [SL353LT](https://cz.mouser.com/ProductDetail/Honeywell/SL353LT?qs=pB3G9VbQXIenMOyKzoQ33g%3D%3D)

ChroustJan, 2020-10-26:
@kaklik neměli by to spíš být dva PCB? Očekávám, že se dopředu bude vědět technologie, která bude použita. Myslím si, že zde potřebujeme mít co nejmenší PCB.

kaklik, 2020-10-26:
> @kaklik neměli by to spíš být dva PCB? Očekávám, že se dopředu bude vědět technologie, která bude použita. Myslím si, že zde potřebujeme mít co nejmenší PCB.

Je to zřejmě možnost.  Podstatné bude hlavně to, jakým způsebem ty PCB budou připevněné a zda budou záménné.

kaklik, 2020-11-12:
Solved by https://github.com/ThunderFly-aerospace/TFPROBE01

closed 2020-11-12 by kaklik