ThunderFly-aerospace/TFRPM01 #8: Zlepšení čitelnosti popisků

kaklik, 2020-12-09:
Popisky na boku krabičky jsou téměř nečitelné. 
Navrhoval bych je udělat vystouplé místo zapuštěných.

kaklik, 2020-12-15:
Zatím čitelnost popisků touto změnou nebyla moc zlepšena. 
je navíc problém s tisknutelností TF loga, které se v současné velikosti s tryskou 0.4 mm netiskne příliš kvalitně.

kaklik, 2020-12-22:
Nakonec byly popisky z krabičky odebrány úplně.  Důvod je, že je zřejmě není možné kvalitně vytisknout a tím působí neseriózně. 
Další vhodnější řešení by zřejmě bylo použití samolepek.  V případě této konstrukce krabičky zřejmě jedna větší zakrývající šroub a celou vrchní stranu krabičky. 
Samolepka tak potřebuje mít zaoblené rohy se stejným rádiusem, jako má krabička.

ChroustJan, 2021-02-11:
Pokud budeme mít správně dlouhý šroub, tak samolepka ze strany matice může být přes celou plochu. Vrchní by měla mít otvor pro šroub.

kaklik, 2021-02-11:
Já do teď uvažoval o samolepce pouze z jedné strany a to ze strany víčka. Protože ta krabička je konstruovaná tak, že se buď na spodní  stranu přilepí suchý zip, kterým se nalepí někam na UAV.  A nebo se použije delší šroub, kterým se někam přišroubuje skrz víčko.  (To má ale komplikaci že vidím poměrně velký problém v tom dělat kulatou díru pro šroub do samolepky.)

ChroustJan, 2021-02-11:
Tak já to bral tak, že ten suchý zip dáš přes tu samolepku a nebo i z druhé strany bude díra pro šroub. Doufám, že tu díru umí udělat ten tiskař. Mohl by jedině být problém, že nebudou umět tak malou. A pak nevím jak to bude s přesností lepení, ale asi by nevadilo mít větší díru v samolepce. Pokud bude samolepka z obou stran, tak to zmenší ten důraz na kvalitu 3D tisku a bude to podle mě celé vypadat lépe. O tomto by jsme se měli pobavit společně před začátkem návrhu samolepek.

kaklik, 2021-03-13:
Problém s kvalitou potisků je pravděpodobně vyřešen návrhem samolepky. 

[image]

ChroustJan, 2021-03-13:
Nápisy z tisknutelné krabičky odstraněny. Jejich funkci zastane samolepka.

closed 2021-03-13 by ChroustJan