ThunderFly-aerospace/TFRPM01 #10: Změna pullupu pro měření s TFPROBE01

roman-dvorak, 2020-12-21:
Změnit hodnotu pullupu na vstupu signálu u TFRPM01

ChroustJan, 2021-02-11:
Aktuálně se testuje 10k

kaklik, 2021-02-18:
S hodnotou 10k byl při použití v kombinaci [se snímacím bubínkem](https://github.com/ThunderFly-aerospace/TF-G2/issues/64#issuecomment-761890227) pozorován problém, kdy na přímém slunečním světle přestalo fungovat čítání snímače. 

Současně existuje předpoklad, že je to způsobeno především prosvícením bubínku slunečním světlem. Je nutné tento problém dále analyzovat. Může však dojít k nutnosti snížit hodnotu pull-upu, aby byla snížena citlivost senzoru.

ChroustJan, 2021-02-18:
Zkusíte tohle nějak analyzovat? Bylo by dobré znát hodnoty do nové várky osazování. A pokud není chtělo by to popsat do návodu, že při použití této technologie může být potřeba doladit hodnoty. Podle zvolené mezery a odrazného materiálu.

kaklik, 2021-02-21:
Tak z aktuálních poznatků vyplývá, že testovaná hodnota pull-upu 4k7 je minimálně pro použití na TF-G2 příliš nízká. Kromě jiného se ukázala i značná citlivost na barvu materiálu. Viz následdující grafy: 

Fialová odrazná plocha
[image]

Modrá odrazná plocha
[image]

Šedá odrazná plocha
[image]

V žádném z těchto případů nedocházelo ke spínání vstupu TFRPM a kombinaci pull-up/materiál nelze použít. Při použití rezistoru 10k je situace poněkud lepší:

kaklik, 2021-02-23:
Tady jsou nové obrázky po následujících změnách:

- zvětšení proudu IR LED na 20 mA zmenšením rezistoru R1 ze 180 na 82 Ohm
- Instalace 10k pull-upu na TFRPM in TFPROBE

Fialová
[image]
Modrá
[image]
Šedá
[image]

kaklik, 2021-03-04:
Navrhoval bych  pull-up R1 zkusit používat 22k.   Omezení proudu rezistorem R2 omezit hodnotou 82 Ohm. 
Upravit je potřeba i [nastavení hodnot na TFPROBE viz samostatné issue](https://github.com/ThunderFly-aerospace/TFPROBE01/issues/5). 

Po těchto úpravách vypadají signály pro fialový a zelený bubínek relativně použitelně.  Systém je ale hodně citlivý na vzdálenost možná zde bude muset dojít ještě k mírným konstrukčním úpravám. 

Fialový bubínek: 
[image]

Zelený bubínek ve stejné vzdálenosti jako fialový: 
[image]

Zelený bubínek při kratší vzdálenosti: 
[image]

Šedivý bubínek je ale stále nepoužitelný:

[image]

ChroustJan, 2021-03-13:
Osazeno R1 22K a R2 82R

kaklik, 2021-03-19:
S nezměněnou konfigurací oproti předchozímu komentáři jsem vyzkoušel bubínek vytištěný z barvy "Teal".

[image]

Zatím to vypadá, že zhlediska snímání je to nejvhodnější barva.

kaklik, 2021-03-21:
Po úpravě  TFPROBE podle https://github.com/ThunderFly-aerospace/TFPROBE01/issues/5

A při zachování konfigurace TFRPM https://github.com/ThunderFly-aerospace/TFRPM01/issues/10#issuecomment-798528538 , jsem zkoušel změřit napěťový průběh na vstupu TFRPM ze sondy TFPROBE. 

### Fialový  klobouček

[image]

### Teal  klobouček

[image]

Jak je z grafů vidět, tak hodnota minimálního i maximálního napětí klesla. S tím že maximální napětí je nyní pod hranicí 2V, což neodpovídá vstupním napěťovým úrovním #16 .   Důsledkem toho, je že u barvy Teal je TFRPM senzor část úvratě v "sepnutém stavu" s rozsvícenou LED a nečítá pulzy. 

Z testovaných barev  (Šedá, modrá, fialová, Teal, Bílá, Zelená), je teď optimální kotouček vytištěný ze zelené barvy, který funguje spolehlivě. 

[image]

kaklik, 2021-03-23:
Jinak bez kloboučku je na tom výstupu TFPROBE  cca 3V. Což znamená, že v té horní mezi je ještě celkem dost velká rezerva, která zaniká tím že ten senzor je částečně osvětlený odraženým světlem z vnitřku kloboučku. 

Pokud by zevnitř bylo něco, co to světlo lépe pohltí, než prázdný prostor, tak by se zřejmě dala snížit ta citlivost na barvu.

kaklik, 2021-08-05:
Záležitost byla nakonec vyřešena přebarvením zubů klobočku viz komentář: https://github.com/ThunderFly-aerospace/TF-G2/issues/64#issuecomment-808915707

Toto přebarvení zřejmě dále umožňuje případně snížit hodnotu pull-upu na TFRPM.

closed 2021-08-05 by kaklik