ThunderFly-aerospace/TFRPM01 #13: Ověřit správnost I2C adresy na potisku PCB

ChroustJan, 2021-02-17:
Ověřit správnost I2C adresy na potisku PCB. Že platí 0x50 v základním nastavení a 0x51 po přepojení.

kaklik, 2021-02-18:
Zde není moc co kontrolovat, nastavená nula na vstupu A0. Nastavuje adresu na 0x50 viz datasheet. 

[image]

Je akorát podivné, že v PX4 firmware je s adresou 0x50 zacházeno jako s dekadickou adresou 80. Což mi přijde poněkud matoucí.. Dělá se to tak v PX4 běžně? @slimonslimon ? 

Každopádně mi ale přijde nepraktické, že propojka nemá označenou adresu, která je zvolena jejím nastavením. Možná to nemusí být nutně přímo v popisku u propojky. Ale měl by tam myslím být odkaz na tabulku z druhé strany, kde je teď však adresa pouze 0x50  

[image]

Myslím že by jako odkaz šla použít ta šipka, kterou symbol letovací propojky na potisku jíž obsahuje. 

[image]

slimonslimon, 2021-02-18:
Těch 80 jsme tam prostě napsal, protože jsem v tom tehdy neviděl moc rozdíl. Žádný specifický důvod v tom není. (původně což možná platí doteď je možné adresu nastavovat parametrem - ten parametr se pak pochopitelně vypisoval dekadicky - celou tuto featuru asi zruším)

ChroustJan, 2021-02-18:
@kaklik důležitý je ten stav ve kterém to je po osazení a budeme to dodávat. Ten kdo do toho zasahuje by měl už něco vědět. Pokud to myslíš tak, že před 0x50 chceš dát symbol toho trojúhelníku tak ok.

Dvě adresy nevím zda tam dávat, aby to nebylo matoucí. A nechce se mi všechny standardně přeškrtávat.

kaklik, 2021-02-18:
> Těch 80 jsme tam prostě napsal, protože jsem v tom tehdy neviděl moc rozdíl. Žádný specifický důvod v tom není. (původně což možná platí doteď je možné adresu nastavovat parametrem - ten parametr se pak pochopitelně vypisoval dekadicky - celou tuto featuru asi zruším)

No to nastavování adresy (nějak) je asi docela důležitý. Protože vzhledem k tomu, že se TFRPM dají řetězit. Tak lidi mají tendenci zapojovat třeba alespoň dva za sebe.  (Už jsme minimálně dotaz na to jestli to jde od jednoho zákazníka měli). Ok ale závěrem zřejmě je, že je celkem zvykem pracovat s dekadickou hodnotou parametrů. Což vede k tomu, aby na PCB byla ideálně vyznačena i dekadická hodnota. Nebo je to zjevné a zákazníci si to převedou? Případně to bude vysvětlené v dokumentaci?  Protože naopak jinde bych rekl, že se s dekadickou formou I²C adresy naopak běžně nepracuje.

slimonslimon, 2021-02-18:
no na PCB bych to uváděl hexadecimálně stradičním 0x na začátku.

ChroustJan, 2021-02-19:
Zkoušel jsem tam dát trojúhelníček a druhou adresu nevypadá to hezky. A nemáme tam moc místa. Navrhuji tam nechat jen 0x50. Pokud si to někdo předělá, tak si to může rozlišit tím, že si tu adresu škrtne. Ty škrtlé nebudou mít 0x50.  Druhá věc je že to chceme prodávat v krabičkách se samolepkou. Napadla mě ta věc, zda ke každé zásilce nedávat samolepku navíc s variantou adresy 0x51. Bylo by to dle návrhu Romana s tím že z výroby by byl křížek u 0x50 a v sáčku by dostali samolepku s křížkem 0x51, pro přelepení. Na každé samolepce by ale byla vždy i druhá adresa aby bylo vidět, že je možné to měnit.

ChroustJan, 2021-02-19:
[image]

kaklik, 2021-02-20:
Aktuálně vyznačená adresa na modulu 0x50 označuje nastavení z výroby. 

Jako alternativní možnost vidím umístění adres nad GND pin hřebínku. 
Dokumentaci v readme jsem ale upravil tak, aby odpovídala stavu prezentovanému v komentáři výše.  https://github.com/ThunderFly-aerospace/TFRPM01/issues/13#issuecomment-781923042

closed 2021-02-20 by kaklik