ThunderFly-aerospace/TFRPM01 #14: Návrh samolepky

ChroustJan, 2021-02-19:
Obsah:
QR kód - buď na thunderfly.cz nebo na repozitář

Zvážit možnost vytvoření dvojice samolepek pro spodní stranu. Jedna se zaškrtnutím adresy 0x51 a druhou s 0x50.

kaklik, 2021-02-19:
Adresa na spodní (montážní) straně je celkem k ničemu.

ChroustJan, 2021-03-13:
Výsledný návrh samolepky:
[image]

closed 2021-03-13 by ChroustJan

roman-dvorak, 2021-03-31:
Mohl by mi prosím někdo (@kaklik  nebo @ChroustJan ) změřit velikost toho archu (nebo říct, co je to za formát), na kterém jsou vytištěné samolepky?

kaklik, 2021-03-31:
> Mohl by mi prosím někdo (@kaklik nebo @ChroustJan ) změřit velikost toho archu (nebo říct, co je to za formát), na kterém jsou vytištěné samolepky?

Ten arch je 303x240mm ale jestli je to pevný formát, to nevím. Okraje vypadají že jsou něčím dodatečně střihané. Na rubové straně je logo téhle firmy: https://www.arlon.com/na_en/