ThunderFly-aerospace/TFRPM01 #15: Proražená dioda D1

kaklik, 2021-02-19:
Na jednom z modulů došlo během relativně běžného používání k proražení zenerovo diody D1.  Hlavní anomálií byla nízká teplota (okolo -12 °C).

[image]

Lze předpokládat, že dioda odešla v důsledku otevření na vyšším napětí s následným nadproudem, který diodu zničil. 
Podle schéma je tam neznámý typ diody s otevíracím napětím 5.6V To je však velmi blízko změřenému napájecímu napětí 5.4 V při napájení z akumulátoru přes CUAV power modul. 
Vzhledem k absenci přesného označení typu diody ve schéma, nelze určit jaká je její teplotní charakteristika a zda na této události mohla mít podíl nízká teplota během provozu.

ChroustJan, 2021-02-21:
@kaklik  BZV55C-5,6V

kaklik, 2021-03-04:
K tomuto problému bylo vytvořeno [issue v oficiální dokumentaci Pixhawk.](https://github.com/pixhawk/Pixhawk-Standards/issues/15#issuecomment-786302167)
Zůstává však bez relevantního řešení.

kaklik, 2021-03-06:
Issue v dokumentaci stále zůstává bez odpovědi. Navrhuji problém zkusit vyřešit použitím typu zenerovo diody s vyšším otevíracím napětím: BZV55C 6.2BSA

ChroustJan, 2021-03-07:
@kaklik  myslíš prostě BZV55C6V2 ? Nenašel jsem co znamená BSA

kaklik, 2021-03-07:
> @kaklik myslíš prostě BZV55C6V2 ? Nenašel jsem co znamená BSA

BSA specifikuje možný rozptyl otevíracího napětí. Viz datasheet například zde: https://www.rectron.com/data_sheets/bzv55c-bs.pdf

ChroustJan, 2021-03-12:
Parametr je to hezký, ale našel jsi místo, kde to jde koupit?

Máme tuto: https://www.tme.eu/cz/details/bzv55c6v2/zenerovy-diody-smd/dc-components/

ChroustJan, 2021-03-13:
Dán tento typ:  https://www.tme.eu/cz/details/bzv55c6v2/zenerovy-diody-smd/dc-components/

closed 2021-03-21 by kaklik