ThunderFly-aerospace/TFRPM01 #16: Vstupní logické úrovně

kaklik, 2021-02-21:
Snažil jsem se teď zjistit, jaké jsou skutečné hodnoty vstupních napěťových úrovní na vstupu IN sondy. 
Zjistil jsem, že to vlastně není vůbec snadné, neboť ve schéma není jasné, na jaké napájecí napětí je připojen smittuv klopný obvod. 

Teprve až po otveření kicadu lze zjistit, že je trvale připojen na 3.3V. 

[image]

Nicméně i to úplně neřeší situaci, neboť v [datasheetu](https://assets.nexperia.com/documents/data-sheet/74HC_HCT1G14.pdf) je asi opomenuta tabulka úrovní pro hodnotu napájení 3.3V. 

I tak je ale rozptyl rozhodovacích úrovní, pro mne nečekaně velký. Obávám se tak, zda z použití této součástky nemohou v budoucnu vzniknout nějaké potíže. Například ve smyslu, že mezi jednotlivými kusy by bylo potřeba mít různé hodnoty pull-up rezistorů

[image]

Zatím ale praktické zkušenosti tomuto rizikovému scénáři příliš nenasvědčují. Nicméně je zřejmě potřeba provést testy na více různých kusech, ideálně i z různých sérií. 

Nyní jsem se pokusil alespoň zviditelnit napájecí nožičku na schématické značce ve schéma. Nicméně jsem si pak všiml, že v repozitáři zřejmě není nahraný poslední návrh. Tím pádem jsem tyto změny neuložil.

ChroustJan, 2021-02-22:
Napájecí napětí je definováno zde:
[image]

kaklik, 2021-02-23:
Na základě #17 navrhuji přejít na obvod [5SN74LVC1G14-Q1](https://www.ti.com/lit/ds/symlink/sn74lvc1g14-q1.pdf), který logické úrovně definuje o něco přesněji.

[image]

kaklik, 2021-02-25:
Pro praktické účely bych nyní navrhoval pracovat s předpokladem, že potřebujeme dosáhnout napěťových úrovní pod 0.58 V pro log 0 a  více jak 2V pro log 1. 

Další pokračování bude v issue #10

closed 2021-02-25 by kaklik