ThunderFly-aerospace/TFRPM01 #17: Zatížení 74HC1G14GW proudem z LED

kaklik, 2021-02-21:
Při řešení  #16 jsem zjistil, že v [datasheetu](https://www.ti.com/lit/ds/symlink/sn74hc14.pdf?ts=1613907697872) se zmiňují o: 
"Recommended output conditions: –  Load currents should not exceed 4 mA per output"

Přibližným odhadem (3,3−1,7)÷180  jsem ale zjistil, že proud skrz LED je nyní zřejmě něco jako  0,0089A. Což zřejmě nesplňuje tento požadavek. 

Je tak na zvážení, zda není vhodné použít jiný typ této součástky.

ChroustJan, 2021-02-22:
@kaklik zkusil by jsi prosím co nejdříve najít vhodnou náhradu?

kaklik, 2021-02-22:
> @kaklik zkusil by jsi prosím co nejdříve najít vhodnou náhradu?

Pracuju na tom. Domnívám se ale, že by bylo zároveň vhodné zjistit, jak moc se sníží jas indikační LED v případě, že se dodrží výše uvedené kritérium zátěže výstupu.

kaklik, 2021-02-23:
Asi bych navrhova začít používat tohle: [SN74LVC1G14-Q1](https://www.ti.com/lit/ds/symlink/sn74lvc1g14-q1.pdf?ts=1614074438985&ref_url=https%253A%252F%252Fwww.ti.com%252Fsitesearch%252Fdocs%252Funiversalsearch.tsp%253FsearchTerm%253Dsingle%2Bschmitt%2Btrigger%2Binverter)

Je to sice hůře dostupné a mírně dražší, ale není to takový mizerný řešení, jako ten předchozí typ. 
Tohle je navíc kvalifikovaný pro automotive a má to vysokou ESD odolnost.

ChroustJan, 2021-02-23:
Vypadá to dobře. Pokud by se chtělo šetřit, tak je ke zvážení verze bez Q1. Ale myslím si, že příplatek 1kč si můžeme dovolit.

ChroustJan, 2021-03-13:
V novém osazování použito IO: SN74LVC1G14-Q1

kaklik, 2021-03-21:
Změřil jsem napěťové úrovně na  výstupu IO U2A a na LED.

### Průběh napětí na výstupu IO

[image]

### Průběh napětí na LED D2 

[image]

Z obou průběhů je vidět, že úbytek na rezistoru R4 s hodnotou 51 Ohm, je

closed 2021-03-21 by kaklik