ThunderFly-aerospace/TFRPM01 #18: Zkosit hranu otvoru pro hlavu šroubu

kaklik, 2021-03-18:
Samolepku technicky není možné na krabičku nalepit úplně přesně.  Je tak vidět okraj otvoru pro hlavu šroubu, který je díky tisku nepřesný. 

[image]

Situaci by zřejmě mohlo vyřešit malé zkosení hrany otvoru, tak aby nepřesná hrana byla schována pod samolepkou.   Otvor zřejmě stačí zkosit o 45 stupňů ve výšce třeba 5ti vrstev (Aktuálně předpokládaná tisková vrstva je 0.1mm). Tím bude model mít mírně větší návrhový otvor než samolepka.  Při tisku na podložku se ale hrana otvoru mírně rozplácne, takže se rozdíl průměrů vyrovná.

ChroustJan, 2021-03-18:
Buď to neřešit, nebo úprava krabičky.

kaklik, 2021-03-20:
I přes to, že se mi samolepku tentokrát na krabičku nepodařilo nalepit tak přesně, tak úprava krabičky je výrazné zlepšení. 

[image]

closed 2021-03-20 by kaklik