ThunderFly-aerospace/TFRPM01 #20: Svítivost LED pod samolepkou

kaklik, 2021-03-18:
Aktuální samolepka je natištěna na bílé PVC fólii.  Indikační LED skrz fólii nesvítí příliš zřetelně:

[image]

Na samolepce je navíc vidět, že nesvítí celá bílá plocha vlivem posunutí samolepky při lepení na krabičku.  Tmavá plocha okolo puntíku ale naopak prosvětlená není, čemuž pomáhá černý lem okolo bílého puntíku. 

Možná řešení
-  Zvětšit proud LED, aby bylo dosaženo maximální svítivosti  
-  Zvětšit otvor pod samolepkou nad průměr puntíku na samolepce
-  Samolepku objednat na průhledné fólii
-  Otvor směrem k LED udělat kónický s průměrem mírně větším, než je průměr otvoru PCB pro LED.

ChroustJan, 2021-03-18:
Za mě to buď neřešit, nebo udělat úpravu krabičky.

ChroustJan, 2021-03-19:
Bude zvýšen proud LED na 24mA. R4 ze 180R bude 51R.

kaklik, 2021-03-20:
Ještě mne napadla jedna možnost, jak to zlepšit - snížit tloušťku víčka. Nyní je konec šroubu při dotažení utopený o 1,45mm pod povrchem krabičky.  Je zde tak 1mm rezerva na přiblížení LED k samolepce.

roman-dvorak, 2021-03-20:
Dobře, snížím tloušťku víčka o 1mm (pokud tomu něco nebude vadit).

Na fotce z prvního komentáře jsem vůbec netušil, že ta dioda svítí. Nová fotka je už s úpravou, kdy do led teče vyšší proud?

Moc se mi nelíbí, jak to svítí i mimo kroužku. Navrhoval bych ho zase trochu zúžit a nechat kuželovitý.

Myslím, že tato úprava neblokuje focení aktuální verze.

kaklik, 2021-03-20:
> Na fotce z prvního komentáře jsem vůbec netušil, že ta dioda svítí. Nová fotka je už s úpravou, kdy do led teče vyšší proud?
>

Ne, to je zatím se starým PCB TFRPM01B. Stejný proud, jako předchozí fotka. 
 
> Moc se mi nelíbí, jak to svítí i mimo kroužku. Navrhoval bych ho zase trochu zúžit a nechat kuželovitý.
> 

Ok, jak myslíš, mě tohle zrovna nijak nevadí.

kaklik, 2021-03-23:
> Dobře, snížím tloušťku víčka o 1mm (pokud tomu něco nebude vadit).

Tohle je zřejmě vhodné udělat i kvůli montáži. Tu normální nerezovou matku se mi před chvílí pod tou samolepkou podařilo skřížit, což je docela nepříjemná situace.  Předpokládám, že nižší díra tomu alespoň trochu zabrání.

roman-dvorak, 2021-03-24:
Víčko ztenčeno o 1 mm v #1519754

kaklik, 2021-03-24:
Pro svítivost LED jsme udělali maximum. Aktuální stav je použitelný. Pro ty, co vyžadují ještě větší svítivost jsem založil[ tenhle návod](https://github.com/ThunderFly-aerospace/TFRPM01/blob/TFRPM01C/doc/README.md).

closed 2021-03-24 by kaklik