ThunderFly-aerospace/TFRPM01 #21: Přesná pozice LED

kaklik, 2021-03-21:
Při testování svítivosti LED #20  jsem přišel na to, že otvor v krabičce zřejmě není přesně na správném místě odpovídajícím otvoru v PCB pro LED. 
Tahle skutečnost, pokud se potvrdí, tak kromě krabičky ovlivňuje ještě i návrh samolepky #14.

roman-dvorak, 2021-03-24:
Ano, vypadá to tak, že je LED skutečně posunutá. 

Takto vypadá posuv na PCB oproti šroubu. 
[image]

Na krabičce/samolepce je to oproti šroubu na druhé straně. (zjistím přesnou hodnotu) 

Tohle bych navrhoval vyřešit posunutím LED na PCB při další výrobě. V okolí LED je relativně místo a připadá mi to jako bezpečná úprava. (Pokud jsou aktuální samolepky vyhovující)

kaklik, 2021-03-24:
Jo souhlasím..  jen tu prosím pak napiš, kam @ChroustJan  má tu LED posunout. Nebo případně, jestli je ten screenshot správně, tak to zřejmě jde rovnou udělat.

roman-dvorak, 2021-03-24:
Ne! ještě to @ChroustJan  nedělej, pošlu přesné umístění.

kaklik, 2021-03-24:
@roman-dvorak  Není náhodou možné, že puntík na samolepce je správně, ale je špatně díra v krabičce? 
Protože je mi podezřelé, že už se mi po několikáté podařilo tu samolepku vůči puntíku pro LED nalepit posunutou na stejnou stranu..

roman-dvorak, 2022-03-17:
Na PCB byla LEDka posunutá na druhou stranu, než měla být. Byla posunuta o přesnou pozici, která vychází z openscadu, ale na druhou stranu. To vytvořilo posuv o více než půl milimetru oproti otvoru.  Kontroloval jsem, že to sedí s openscadem i se svg samolepky. 

Zároveň jsem tam změnil footprint pro LEDku z knihoven z Kicadu - mělo by to usnadnit osazování. 

[image]

Opraveno ve verzi D

ChroustJan, 2022-03-21:
@roman-dvorak přijde mi, že naše LED do zvoleného pouzdra nepasuje :(
[image]

ChroustJan, 2022-07-20:
Provedl jsem opravu pouzdra LED, ale bohužel jsem posunul pozici LED. Je potřeba ohlídat správnou pozici. @roman-dvorak není možný nějaký posun, když bude docházet k tisku nových samolepek a výrobě nové krabičky?

kaklik, 2022-09-28:
Tohle je teď v jakém stavu? Je to v nové revizi D už vyřešené, nebo není?

ChroustJan, 2022-09-29:
Pro verzi D jsem navrhl nové frézované pouzdro. Pozice jsou nové po dohodě s @roman-dvorak.

roman-dvorak, 2022-09-29:
Jo, vzhledem k tomu, že tam ledky budou dvě, bude úprava v krabičce a
samolepce, tak tohle nemá smysl řešit.

Nové pozice jsou umístěny tak, že je lze snadno odměřit od středu šroubu.

closed 2022-09-29 by roman-dvorak