ThunderFly-aerospace/TFRPM01 #24: Výkres

kaklik, 2021-03-28:
Chtěl jsem ověřit jak lze vytvořit výkres ve FreeCADu a při té příležitosti jsem zjistil, že v souboru TFRPM01C.wrl jsou špatná data.

[image]

ChroustJan, 2021-03-29:
Nedalo by se tohle generovat nějak automaticky? Aktuálně jsem nějak rychle nepřišel nato, jak to vygenerovat nově.

kaklik, 2021-03-29:
> Nedalo by se tohle generovat nějak automaticky? Aktuálně jsem nějak rychle nepřišel nato, jak to vygenerovat nově.

Je to jeden z formátů, který lze ručně importovat z KiCAD PCB prohlížeče. Jestli to jde exportovat nějak automaticky je zřejmě spíše otázka na @roman-dvorak .

roman-dvorak, 2021-03-29:
Do doby, kdy se používal starý formát, tak bych řekl docela jistě ano :) Ale teď na to neumím odpovědět. Je potřeba zjistit chyby, co tam aktuálně jsou. I tohle generování se nyní provádí nějak jinak.

ChroustJan, 2021-03-29:
Tohle mi přijde vhodné generovat automaticky, protože na aktualizaci se dá snadno zapomenout. Dává mi dost zabrat udržet schéma aktuální.

ChroustJan, 2021-08-25:
Vygenerování nového modelu:
[image]

closed 2021-08-25 by ChroustJan