ThunderFly-aerospace/TFRPM01 #27: Optical probe readout

laronll, 2021-11-21:
I would like to know if the optical probe needs to have any code change for it to work? Another question is that aiming the board perpendicular to a propeller with attached reflective material should make the sensor get readings? And if I am looking for RPM readings I should be looking for the frequency that the code prints out?

Thank you in advance

roman-dvorak, 2021-11-21:
Hi, I assume that you want to use it with autopilot with PX4 firmware? 

There should be nothing to modify anything. The SW should be ready for such a measurement. Number of reflective marks (per 1 rotation) can be set through [PX4 parameters](http://docs.px4.io/master/en/advanced_config/parameter_reference.html#PCF8583_ADDR). The driver provides value as frequency (count of revolutions per second, in Hz). Frequency is logged by autopilot and forwarded through MavLink network. 

[image]

In addition to the sensor (frequency sensor) TFRPM01, we also have a probe [TFPROBE02](https://github.com/ThunderFly-aerospace/TFPROBE01) (on the picture) that contains the optical reflective sensor itself. The sensor used works at a small distance (units of mm) so I think it is better to measure the engine speed than the propeller. (The reflective material on the propeller can change the position of the centre of gravity quite significantly.) Or create your custom probe that will detect reflections from the propeller at a bigger distance. The TFRPM on the input contains a pullup resistor.

roman-dvorak, 2022-03-17:
@laronll  Do you have any additional questions? Otherwise, I would close this issue.

closed 2022-09-19 by kaklik