ThunderFly-aerospace/TFRPM01 #28: Power and #INT LED

roman-dvorak, 2021-12-28:
Modulu chybí ledky signalizující napájení a signál z INT pinu.

Na INT signálu je například poznat, jestli IO je zinicializovaný nebo by přes něj šlo blikat po načtení několika vstupních signálů tak, aby to bylo viditelné okem i při provozních otáčkách rotoru.

roman-dvorak, 2022-03-17:
Interrupt ledka při neinicializaci nebude blikat (jak bylo zamýšleno), protože nemá zdroj hodin. Takže totu LEDku bych tam v dalšich verzích nedával. 

Je potřeba se rozhodnout, jestli nepřidat power LEDku.

kaklik, 2022-03-17:
> Interrupt ledka při neinicializaci nebude blikat (jak bylo zamýšleno), protože nemá zdroj hodin. Takže totu LEDku bych tam v dalšich verzích nedával.

A jde s ní blikat aspoň softwarem, nebo ten INT signál nelze vůbec ovládat přes I2C? 

> Je potřeba se rozhodnout, jestli nepřidat power LEDku.

Za mě je to dobrý nápad, pokud to technicky lze a nezvětší nároky na PCB. Nemusí ale být reverse mount.

kaklik, 2022-09-28:
Zdá se že ve schéma TFRPM01D je tato LED implementována ve formě zelené reverse mount LED. Chybí u ní však hodnota rezistoru. Není tak jasné, jak moc bude v konečném provedení svítit. 

[image]

closed 2023-02-08 by kaklik